ros2-rust/ros2_rust #371: Failed to build depends on `builtin_interfaces`

sprhawk, 2024-03-14:
https://github.com/ros2-rust/ros2_rust/blob/27bc35584badf0406927b5d6c88625de2ef52580/rclrs/Cargo.toml#L29

I got failure building on `builtin_interfaces`. I [removed it ](https://github.com/iotpi/ros2_rust/commit/1399660123fac828cc82c1fd8fc5ee7408fbd2cd) then it is built successfully.

I'm not sure why it is a dependency of rclrs.

sprhawk, 2024-03-14:
I read the [building.md](https://github.com/ros2-rust/ros2_rust/blob/main/docs/building.md) again

maspe36, 2024-03-14:
> I got failure building on builtin_interfaces.

Can you share this failure and maybe your workspace folder structure? I have not seen this error myself

sprhawk, 2024-03-15:
> > I got failure building on builtin_interfaces.
> 
> Can you share this failure and maybe your workspace folder structure? I have not seen this error myself

I built whole ROS2 myself in Debian Bookworm

I tried later, more libs like `libstatistics_collector` didn't get rust subdir built.

mxgrey, 2024-03-15:
The vendored version of `builtin_interfaces` that you're using in your diff was a workaround from early in the rclrs development that shouldn't be needed anymore.

It sounds like you're trying to build the `ros2_rust` repo in a single workspace with all the other critical packages of ROS 2. If that's the case you'll need #370 which I opened recently to make that possible. I recommend checking out that branch and doing a clean build (run `$ rm -rf build install` before building).

sprhawk, 2024-03-16:
> The vendored version of `builtin_interfaces` that you're using in your diff was a workaround from early in the development of rclrs that shouldn't be needed anymore.
> 
> It sounds like you're trying to build the `ros2_rust` repo in a single workspace with all the other critical packages of ROS 2. If that's the case you'll need #370 which I opened recently to make that possible. I recommend checking out that branch and doing a clean build (run `$ rm -rf build install` before building).

Thanks I tried your branch but not clean all build and install, I'll try it again.

maspe36, 2024-03-20:
I am not seeing the original problem, but the second problem you describe,

> I tried later, more libs like libstatistics_collector didn't get rust subdir built.

I do see. I don't always see it, but when it pops up in my workspace I need to do a clean build to resolve it. Next time my workspace is borked I'll take a closer look.

mxgrey, 2024-03-20:
I think once #380 is merged we can close this issue.

sprhawk, 2024-03-26:
> #380 has been merged in.
> 
> @sprhawk, would you be able to try this again and let us know if the issue is resolved?

Thanks, I'll try it before next week.

sprhawk, 2024-03-30:
Awesome ! It is working now ! 
Thanks!

closed 2024-03-30 by sprhawk