ros2-rust/ros2_rust #375: Your own installation process crashes

Guelakais, 2024-03-19:
According to the installation process shown in your Readme, these were the last steps needed to complete the installation:
```
git clone https://github.com/ros2-rust/ros2_rust.git src/ros2_rust
vcs import src < src/ros2_rust/ros2_rust_humble.repos
. /opt/ros/humble/setup.sh
colcon build
```
The `colcon build` produced the following error:
```
[0.978s] WARNING:colcon.colcon_core.verb:No task extension to 'build' a 'ros.ament_cargo' package
[1.587s] WARNING:colcon.colcon_core.verb:No task extension to 'build' a 'ros.ament_cargo' package
[1.659s] WARNING:colcon.colcon_core.verb:No task extension to 'build' a 'ros.ament_cargo' package
[1.683s] WARNING:colcon.colcon_core.verb:No task extension to 'build' a 'ros.ament_cargo' package
[1.711s] WARNING:colcon.colcon_core.verb:No task extension to 'build' a 'ros.ament_cargo' package
[1.762s] WARNING:colcon.colcon_core.verb:No task extension to 'build' a 'ros.ament_cargo' package
[1.791s] WARNING:colcon.colcon_core.verb:No task extension to 'build' a 'ros.ament_cargo' package
Starting >>> rosidl_generator_rs
[1.990s] ERROR:colcon.colcon_cmake.task.cmake.build:Failed to find the following files:
- /ros_ws/install/rosidl_runtime_rs/share/rosidl_runtime_rs/package.sh
Check that the following packages have been built:
- rosidl_runtime_rs
Failed   <<< rosidl_generator_rs [0.04s, exited with code 1]
Starting >>> test_interface_files
Starting >>> entrez
Starting >>> fasta_file_shrinker
Aborted  <<< test_interface_files [0.04s]
```
As an alternative I have colcon another rust package - which colcon can now build just like that, even without `package.xml`.
```
colcon build --packages-select entrez
Starting >>> entrez
--- stderr: entrez
warning: unused variable: `pm_parsed`
  --> src/main.rs:23:13
   |
23 |         let pm_parsed = PubmedArticleSet::read(&pm_xml);
   |             ^^^^^^^^^ help: if this is intentional, prefix it with an underscore: `_pm_parsed`
   |
   = note: `#[warn(unused_variables)]` on by default

---
Finished <<< entrez [3.62s]

Summary: 1 package finished [4.32s]
  1 package had stderr output: entrez
```
As you can see, it worked.

mxgrey, 2024-03-19:
I recommend double-checking that you've installed all these colcon related dependencies and then do a clean build (run `$ rm -rf build install` in the root of your colcon workspace):
```
cargo install --debug cargo-ament-build  # --debug is faster to install
pip install git+https://github.com/colcon/colcon-cargo.git
pip install git+https://github.com/colcon/colcon-ros-cargo.git
```

Guelakais, 2024-03-19:
The dependencies were already fulfilled - I would be very surprised if not if I thought about how long I had been working on the dev container. The cleanbuild actually helped.

closed 2024-03-19 by Guelakais