ros2-rust/ros2_rust #376: Strange Warning by initial rclrs compilation

Guelakais, 2024-03-19:
This happened to me when I ran through the standard installation process. Is this normal?
```
CMake Warning at /ros_ws/install/rosidl_generator_rs/share/rosidl_generator_rs/cmake/rosidl_generator_rs_generate_interfaces.cmake:46 (message):
  Rust actions generation is not implemented
Call Stack (most recent call first):
  /opt/ros/humble/share/ament_cmake_core/cmake/core/ament_execute_extensions.cmake:48 (include)
  /opt/ros/humble/share/rosidl_cmake/cmake/rosidl_generate_interfaces.cmake:286 (ament_execute_extensions)
  CMakeLists.txt:16 (rosidl_generate_interfaces)
```

maspe36, 2024-03-20:
This is a warning we produce in the rosidl_generator_rs package ([here](https://github.com/ros2-rust/ros2_rust/blob/main/rosidl_generator_rs/cmake/rosidl_generator_rs_generate_interfaces.cmake#L46)). 

As the message implies, we do not have action support currently, so we don't generate code for files that live under the `action` folder in an interface package.

esteve, 2024-03-20:
Actions are being tackled in https://github.com/ros2-rust/ros2_rust/issues/244

closed 2024-03-20 by esteve