ros2-rust/ros2_rust #378: Your tutorial is actually too complicated for beginners.

Guelakais, 2024-03-20:
Writing a republisher as the first task is not as easy as it sounds. It requires a lot of difficult tasks to even get a subscription done. Writing a node that subscribes and publishes as the first task - also a middleman in your data pipeline - is like inventing telephone distribution before telephones. In this way, your tutorial does not even address the experienced ros2 user, but also the experienced rust user.  I really think you should simplify this tutorial. Ros2 and Rust are both exciting and work well together. The tutorial is actually a bit daunting. I'll see if I can tweak it a bit to make it a little easier to find your way around ros2 rust. The current installation process is also very confusing in my opinion.

esteve, 2024-03-20:
@Guelakais thanks for trying `ros2-rust`, it's unfortunate your experience with it hasn't been the most pleasant. Would you be so kind to convert this issue into a list of action items we can address?

We would really appreciate it and it'd hopefully help us produce better documentation for our users. Unfortunately the way it's phrased doesn't give us a good idea of what exactly can be improved and what its shortcomings are. Thanks.

esteve, 2024-03-20:
@Guelakais 

> We both know I'm not very good at Github

To be honest, I don't, so I just asked in the hopes we can make the documentaion better for you and for other users :slightly_smiling_face: 

> Look, this is the source code for a simple ros2 rust publis that I've written. Be inspired by it or not.

Thanks! 

This looks rather similar to the example node we have in the tutorial, is there anything you'd like to see changed in this document?

https://github.com/ros2-rust/ros2_rust/blob/main/docs/writing-your-first-rclrs-node.md

Would you prefer a node that doesn't subscribe to a topic and only publishes a message periodically via a timer?

Guelakais, 2024-03-20:
The two nodes are essentially different. The node from your tutorial is the middleman of a data pipeline, the node I show is a pure publisher. This simplifies things enormously, as the node can show within itself whether it works or not. With the node from the tutorial, you first have to start another publisher, which then controls your node. What can be improved is to add logging or terminal output from the node itself, which provides information about the functionality, as can be seen [here](https://docs.ros.org/en/humble/Tutorials/Beginner-Client-Libraries/Writing-A-Simple-Py-Publisher-And-Subscriber.html).  I understand that the tutorial is intended to kill two birds with one stone by showing a node that directly fulfils several functions. Some people want to run before they can walk, others the other way round. In fact, the approach from the official ros2 documentation makes a lot of sense. So first a publisher, then a subscriber. Your republisher can then be moved to the examples or to the 2nd page of the tutorial. From my experience, you mainly write the ROS2 nodes that are at the outer end of the tech stack - especially the drivers - yourself. and these are usually either publishers or subscribers, which then communicate with certain SDKs. They are also usually easier to understand for beginners.
At the moment I'm still working on the documentation for the [node](https://gitlab.com/rust_projects3505446/simple_ros2_node).

esteve, 2024-03-20:
@Guelakais ah, I see what you mean, thanks for elaborating. Yeah, I agree with you, a simple "hello world" project with a publisher and a subscription like the ones we have in the tutorials for `rclcpp` and `rclpy` would be great.

PR #390 (2024-06-28): Add simple talk/listener tutorial, matching the official ROS 2 documentation #378
I've decided to write my own Tutorial for your Repository. Suggestions are welcome!
fixes #378
closed 2024-06-28 by esteve