ros2-rust/ros2_rust #391: QoS for Clients and Services

samouwow, 2024-04-15:
The API for creating new clients and services ([create_client()](https://docs.rs/rclrs/latest/rclrs/struct.Node.html#method.create_client) and [create_service()](https://docs.rs/rclrs/latest/rclrs/struct.Node.html#method.create_service)) doesn't allow for specifying non-default QoS.

From what I can tell, adding the ability to specify QoS should be as simple as passing a QosProfile structure through to the client_options / service_options parameter of the rcl_init functions.

I'm happy to open a PR to add this feature but before I do, I wanted to ask:

1. Is there a reason create_client() and create_service() don't currently allow for a QoS profile?
2. It seems the cleanest way to add this feature will be to modify the existing node function signatures, is there as standard way to make breaking changes like this? (E.g. just go ahead, make a new function and deprecate the old one?)

Guelakais, 2024-04-17:
I had the same problem. You can simply create your own QoSprofile or modify an existing QosProfile. This then works as follows:
```
fn new(context: &Context) -> Result<Self, RclrsError> {
        let node = create_node(context, "simple_publisher").unwrap();
        let _publisher = node
            .create_publisher("publish_hello", QOS_PROFILE_DEFAULT.keep_last(5))
            .unwrap();
        Ok(Self { node, _publisher })
    }
```
In this example, the QoSprofile que depth was adjusted using the keeplast() method. This is exactly how you can adjust any value of an existing QoSprofile. Unfortunately, I only found the methods in [github](https://github.com/ros2-rust/ros2_rust/blob/main/rclrs/src/qos.rs). If you have any more questions about the Qosprofile implementation of rclrs, just ask me. I am currently working with it.

mxgrey, 2024-04-17:
I think custom QoS is discouraged for clients and services because there's a significant risk that they'll misbehave if you give them anything other than reliable quality. 

That being said I think it could be reasonable to customize some things like history depth. I'm not sure what the default service setting is for depth, but I can imagine a very busy service could get overwhelmed if that value isn't high enough.

imdexsam, 2024-05-20:
@Guelakais thanks for the example, are you suggesting I use a standard publisher instead of a client as a workaround for custom QoS?

> Ah. Hang on, wait a minute. Could it be that you are not allowed to create the QosProfile for services? In the [Python](https://docs.ros.org/en/humble/Tutorials/Beginner-Client-Libraries/Writing-A-Simple-Py-Service-And-Client.html) and the [cpp](https://docs.ros.org/en/humble/Tutorials/Beginner-Client-Libraries/Writing-A-Simple-Cpp-Service-And-Client.html) API for ros2 this is not possible either.

It seems it's possible, just hidden behind default arguments (for both [rclpy](https://docs.ros2.org/latest/api/rclpy/api/node.html#rclpy.node.Node.create_client) and [rclcpp](https://docs.ros.org/en/rolling/p/rclcpp/generated/classrclcpp_1_1Node.html#_CPPv4I0EN6rclcpp4Node13create_clientEN6rclcpp6ClientI8ServiceTE9SharedPtrERKNSt6stringERKN6rclcpp3QoSEN6rclcpp13CallbackGroup9SharedPtrE)).

> That being said I think it could be reasonable to customize some things like history depth. I'm not sure what the default service setting is for depth, but I can imagine a very busy service could get overwhelmed if that value isn't high enough.

@mxgrey I think customising history is a good example. It's currently set via the [rcl_client_get_default_options()](https://github.com/ros2-rust/ros2_rust/blob/ede22915de953513a9665881e44a4d56edb453c4/rclrs/src/client.rs#L102) function, which uses [rmw_qos_profile_services_default](https://docs.ros.org/en/humble/p/rcl/generated/function_client_8h_1ab0cab7cc0127b8e5f5cc122e255f1d33.html) (reliable, history 10). My specific use case uses a service to change a closed-loop set point in a micro-controller, which then echos back a confirmation. Given the resource constrained environment, Best_Effort gives a significant performance improvement over Reliable, hence my need to customise the QoS.

As an additional argument in support of setting custom QoS, rclrs already has [QOS_PROFILE_SERVICES_DEFAULT](https://docs.rs/rclrs/latest/rclrs/constant.QOS_PROFILE_SERVICES_DEFAULT.html). If the [create_client()](https://docs.rs/rclrs/latest/rclrs/struct.Node.html#method.create_client) function was updated to better align with [create_subscription()](https://docs.rs/rclrs/latest/rclrs/struct.Node.html#method.create_subscription) (and the same for servers / publishers), users would simply need to pass this existing default QoS in to get the current behaviour.

samouwow, 2024-05-20:
@Guelakais thanks for the example, are you suggesting I use a standard publisher instead of a client as a workaround for custom QoS?

Ah. Hang on, wait a minute. Could it be that you are not allowed to create the QosProfile for services? In the [Python](https://docs.ros.org/en/humble/Tutorials/Beginner-Client-Libraries/Writing-A-Simple-Py-Service-And-Client.html) and the [cpp](https://docs.ros.org/en/humble/Tutorials/Beginner-Client-Libraries/Writing-A-Simple-Cpp-Service-And-Client.html) API for ros2 this is not possible either.

It seems it's possible, just hidden behind default arguments (for both [rclpy](https://docs.ros2.org/latest/api/rclpy/api/node.html#rclpy.node.Node.create_client) and [rclcpp](https://docs.ros.org/en/rolling/p/rclcpp/generated/classrclcpp_1_1Node.html#_CPPv4I0EN6rclcpp4Node13create_clientEN6rclcpp6ClientI8ServiceTE9SharedPtrERKNSt6stringERKN6rclcpp3QoSEN6rclcpp13CallbackGroup9SharedPtrE)).

That being said I think it could be reasonable to customize some things like history depth. I'm not sure what the default service setting is for depth, but I can imagine a very busy service could get overwhelmed if that value isn't high enough.

@mxgrey I think customising history is a good example. It's currently set via the [rcl_client_get_default_options()](https://github.com/ros2-rust/ros2_rust/blob/ede22915de953513a9665881e44a4d56edb453c4/rclrs/src/client.rs#L102) function, which uses [rmw_qos_profile_services_default](https://docs.ros.org/en/humble/p/rcl/generated/function_client_8h_1ab0cab7cc0127b8e5f5cc122e255f1d33.html) (reliable, history 10). My specific use case uses a service to change a closed-loop set point in a micro-controller, which then echos back a confirmation. Given the resource constrained environment, Best_Effort gives a significant performance improvement over Reliable, hence my need to customise the QoS.

As an additional argument in support of setting custom QoS, rclrs already has [QOS_PROFILE_SERVICES_DEFAULT](https://docs.rs/rclrs/latest/rclrs/constant.QOS_PROFILE_SERVICES_DEFAULT.html). If the [create_client()](https://docs.rs/rclrs/latest/rclrs/struct.Node.html#method.create_client) function was updated to better align with [create_subscription()](https://docs.rs/rclrs/latest/rclrs/struct.Node.html#method.create_subscription) (and the same for servers / publishers), users would simply need to pass this existing default QoS in to get the current behaviour.

Guelakais, 2024-05-20:
Ros2 publishers and subscribers are different from clients and services. Currently in [rclrs](https://www.google.com/url?sa=t&source=web&rct=j&opi=89978449&url=https://docs.rs/rclrs&ved=2ahUKEwiG0ZPW-ZuGAxUj1gIHHdfqBP4QFnoECAYQAQ&usg=AOvVaw0Op9POHyBAn_s23B4d5Pb3), custom QoS profiles are only allowed for [publishers](https://docs.rs/rclrs/latest/rclrs/struct.Publisher.html) and [subscribers](https://docs.rs/rclrs/latest/rclrs/struct.Subscription.html). Not for [clients](https://docs.rs/rclrs/latest/rclrs/struct.Client.html) or [services](https://docs.rs/rclrs/latest/rclrs/struct.Service.html). I won't recommend changing the currently implemented methods for creating services or clients, because the underlying struct itself only uses the [nodehandle](https://github.com/ros2-rust/ros2_rust/blob/main/rclrs/src/node.rs) struct, which in turn only uses predefined links from the [`rcl_bindings`](https://github.com/ros2-rust/ros2_rust/blob/main/rclrs/src/rcl_bindings.rs). Individual access by rclrs clients and services to the already implemented [QoS profiles](https://github.com/ros2-rust/ros2_rust/blob/main/rclrs/src/qos.rs) is not yet implemented at all. 
At the moment there is simply a lack of a proper implementation, which would be necessary to be able to assign a custom QoS to a client or service at all.

samouwow, 2024-05-20:
@Guelakais I believe I've achieved it with only minor changes to the rclrs client/server code. All that really needed to happen was to connect an API to the rcl QoS struct used to create the client/server.

Would you mind taking a peek at the PR and let me know how that aligns with your comment?

Guelakais, 2024-05-21:
We will see. :)