ros2-rust/ros2_rust #395: Crate imports_granularity requires nightly

maspe36, 2024-04-29:
This was identified by @luca-della-vedova [here](https://github.com/ros2-rust/ros2_rust/pull/342/files#r1582706665). 

Currently, our CI doesn't actually check that our imports are formatted with crate level import granularity. Additionally, this functionality is an unstable feature (https://github.com/rust-lang/rustfmt/issues/4991) so it's only supported in nightly. 

Running locally without a nightly toolchain
```
ros2_rust/rclrs$ cargo fmt
Warning: can't set `imports_granularity = Crate`, unstable features are only available in nightly channel.
```

The way I see it, we have two options.
1. We manually check imports and try to catch this in PRs
2. We start using a nightly toolchain in our CI. We can mitigate some of the churn for nightly here by selecting a version ourselves.

Guelakais, 2024-04-30:
Isn't this something that should be mentioned directly in the readmy or .rustfmt.toml as a comment? The Cargo manifest format allows comments starting with `#`. Isn't this something that should be mentioned directly in the readmy or .rustfmt.toml as a comment? The Cargo manifest format allows comments starting with `#`. I could do a PR to adapt the README.md.

esteve, 2024-04-30:
@maspe36 I prefer to only target stable for builds, however we could target nightly for formatting, but I wonder how much of a moving target it is.

closed 2024-07-13 by maspe36