ros2-rust/ros2_rust #397: Action_msgs not findable, because of relocation from `/ros_ws/install/action_msgs/share/action_msgs/rust` to `/ros_ws/install/share/action_msgs/rust/`

Guelakais, 2024-05-02:
Since this error is relatively new, I wanted to show it. So colcon tries to find the `action_msgs` when building my package and searches incorrectly.
```
--- stderr: simple_ros_rs                                                                                                                                                                                             
error: failed to load source for dependency `action_msgs`

Caused by:
  Unable to update /ros_ws/install/action_msgs/share/action_msgs/rust

Caused by:
  failed to read `/ros_ws/install/action_msgs/share/action_msgs/rust/Cargo.toml`

Caused by:
  No such file or directory (os error 2)
---
```
As you can see, it looks for the library in the `/ros_ws/install/action_msgs/share/action_msgs/rust` path, where it can't find it. Actually, the library is in the path `/ros_ws/install/share/action_msgs/rust/`.
The comand, with whicht I'm building my packages is `colcon build --continue-on-error --build-base /ros_ws/build --install-base /ros_ws/install --base-paths /ros_default /ros_ws /microros_ws --merge-install`

Guelakais, 2024-05-02:
So actually the colcon flag `--merge-install` makes the dependencies appear in `install/share/` folder instead of directly in `/install` folder, which leads to confusion when building. If I see this correctly, the whole thing is rather a bug of `cargo-ament-build` than of colcon directly.

esteve, 2024-05-02:
This is intended, Rust source files are installed into the share directory. Did you run this on a clean workspace or did you reuse an existing one?

Guelakais, 2024-05-02:
That was a clean installation. I know now how the `--merge-install` flag works. I just didn't realise that `-cargo-ament-build` couldn't handle it.