ros2-rust/ros2_rust #399: Begin Jazzy Testing

maspe36, 2024-05-13:
The [docker images](https://hub.docker.com/_/ros/tags?page=&page_size=&ordering=&name=jazzy) for Jazzy have already been released, so we should be able to start testing things in our actions CI.

maspe36, 2024-05-13:
Draft PR is here 
https://github.com/ros2-rust/ros2_rust/pull/400

maspe36, 2024-05-13:
It is currently failing with

```
Run sudo pip3 install git+https://github.com/colcon/colcon-cargo.git
error: externally-managed-environment

× This environment is externally managed
╰─> To install Python packages system-wide, try apt install
    python3-xyz, where xyz is the package you are trying to
    install.
    
    If you wish to install a non-Debian-packaged Python package,
    create a virtual environment using python3 -m venv path/to/venv.
    Then use path/to/venv/bin/python and path/to/venv/bin/pip. Make
    sure you have python3-full installed.
    
    If you wish to install a non-Debian packaged Python application,
    it may be easiest to use pipx install xyz, which will manage a
    virtual environment for you. Make sure you have pipx installed.
    
    See /usr/share/doc/python3.12/README.venv for more information.

note: If you believe this is a mistake, please contact your Python installation or OS distribution provider. You can override this, at the risk of breaking your Python installation or OS, by passing --break-system-packages.
hint: See PEP 668 for the detailed specification.
```

Seems to be a new thing as of Python 3.11. Now with Jazzy targeting 24.04, we will need to figure out how we want to handle this... probably change our README instructions as well.

Not sure why Rolling is still on 3.10, guess its still based on Ubuntu 22.04.

BrandonSimoncic, 2025-02-28:
I had a similar issue, using pipX leads to further bugs with Colcon. actually just using `--break-system-packages` works for now.

maspe36, 2025-08-09:
Yeah for now `--break-system-packages` is the path we've been using. Once `colcon-cargo` and `colcon-ros-cargo` are released on the build farm they should be apt installable removing the need for `--break-system-packages` with pip.

closed 2025-08-09 by maspe36