ros2-rust/ros2_rust #412: `impl std::error::Error` for parameter errors

alluring-mushroom, 2024-08-02:
Hello :smiley: 

I have been testing this excellent project for work.
In using [eyre](https://docs.rs/eyre/latest/eyre/index.html) for much of my error handling, I have found that the errors implemented in [paramaters.rs](https://github.com/ros2-rust/ros2_rust/blob/e485b1c90893bedab32ee9e54ff5e47c4f6bd004/rclrs/src/parameter.rs) do not implement [std::error::Error](https://doc.rust-lang.org/std/error/trait.Error.html), such as `DeclarationError`:
https://github.com/ros2-rust/ros2_rust/blob/e485b1c90893bedab32ee9e54ff5e47c4f6bd004/rclrs/src/parameter.rs#L629

Would you accept a PR containing these?

I assume you do not wish to depend on [thiserror](https://docs.rs/thiserror/latest/thiserror/) as you haven't used it for [RclrsError](https://github.com/ros2-rust/ros2_rust/blob/e485b1c90893bedab32ee9e54ff5e47c4f6bd004/rclrs/src/error.rs#L76). Is there any advice on the [Display](https://doc.rust-lang.org/std/fmt/trait.Display.html) impls for any of these? I think many of the comments should suffice, such this one:
https://github.com/ros2-rust/ros2_rust/blob/e485b1c90893bedab32ee9e54ff5e47c4f6bd004/rclrs/src/parameter.rs#L630

Thank you for your time.

jhdcs, 2024-08-02:
Hello!

Glad you are enjoying `rclrs` so far, and I hope it proves useful for your work! And thank you for the suggestion - yes, I think we probably would accept a PR for implementing `std::error::Error`.

I think that using the the comments would be a good start for the Display implementations for the errors. However, as you're doing so, think about whether that `Display` implementation would help you figure out what might be going wrong with your program. If you think what the comments say aren't enough to help steer an end user (such as yourself or your work colleagues), it may mean that we need to expand what's `Display`ed during an error.

If anything is unclear, or confusing, please feel free to reach out to us! We will do our best to help. And once again, thank you for the suggestion, and offering to help fix it! It means a **lot** to us!

jhdcs, 2024-08-14:
Discussions about a PR are usually best to do on the PR in question, so I think these questions would be better answered over there.

luca-della-vedova, 2024-10-11:
This has been fixed by https://github.com/ros2-rust/ros2_rust/pull/413 and can be closed

closed 2024-10-12 by maspe36