ros2-rust/ros2_rust #424: No rust crate created for service_msgs during build

azerupi, 2024-10-29:
Hi,
I'm trying to build ros2_rust on jazzy. I understand this is not fully supported yet?

When building I can see that all `*_msgs` packages get a rust crate except for the `service_msgs` package. This then fails the build because it is required by `test_msgs` in `rclrs`.

Does anyone know if this is a jazzy issue or if it is something else?

```
colcon build
Starting >>> builtin_interfaces
Starting >>> unique_identifier_msgs
Starting >>> rosidl_runtime_rs
Starting >>> test_interface_files
Starting >>> latency_dashboard
Finished <<< rosidl_runtime_rs [0.15s]
Finished <<< test_interface_files [0.15s]
Starting >>> rosidl_generator_rs
Finished <<< latency_dashboard [0.17s]
Finished <<< rosidl_generator_rs [0.44s]
Finished <<< unique_identifier_msgs [0.62s]
Finished <<< builtin_interfaces [0.67s]
Starting >>> service_msgs
Finished <<< service_msgs [0.56s]
Starting >>> action_msgs
Starting >>> type_description_interfaces
Finished <<< action_msgs [1.00s]
Starting >>> rosidl_default_generators
Starting >>> rosidl_default_runtime
Finished <<< rosidl_default_generators [0.19s]
Finished <<< rosidl_default_runtime [0.20s]
Starting >>> std_msgs
Starting >>> px4_msgs
Starting >>> rcl_interfaces
Starting >>> rosgraph_msgs
Starting >>> test_msgs
Starting >>> example_interfaces
Starting >>> rclrs_example_msgs
Starting >>> std_srvs
Starting >>> lifecycle_msgs
Starting >>> statistics_msgs
Finished <<< type_description_interfaces [1.24s]
Finished <<< rosgraph_msgs [0.94s]
Finished <<< std_srvs [1.00s]
Finished <<< statistics_msgs [1.04s]
Finished <<< rclrs_example_msgs [1.10s]
Finished <<< lifecycle_msgs [1.49s]
Finished <<< test_msgs [1.92s]
Finished <<< rcl_interfaces [1.96s]
Starting >>> rclrs
Starting >>> composition_interfaces
Finished <<< std_msgs [2.51s]
Starting >>> geometry_msgs
Starting >>> actionlib_msgs
Finished <<< example_interfaces [2.57s]
Finished <<< composition_interfaces [0.80s]
Finished <<< actionlib_msgs [0.71s]
--- stderr: rclrs
error: no matching package named `service_msgs` found
location searched: registry `crates-io`
required by package `test_msgs v1.2.1 (/home/azerupi/work/code/gen3/ros2-dev/install/test_msgs/share/test_msgs/rust)`
    ... which satisfies dependency `test_msgs = "*"` of package `rclrs v0.4.1 (/home/azerupi/work/code/gen3/ros2-dev/src/ros2_rust/rclrs)`
---
Failed   <<< rclrs [1.95s, exited with code 1]
```

```
fd -p '(.+)_msgs/rust$' install

install/action_msgs/share/action_msgs/rust/
install/actionlib_msgs/share/actionlib_msgs/rust/
install/diagnostic_msgs/share/diagnostic_msgs/rust/
install/geometry_msgs/share/geometry_msgs/rust/
install/lifecycle_msgs/share/lifecycle_msgs/rust/
install/nav_msgs/share/nav_msgs/rust/
install/px4_msgs/share/px4_msgs/rust/
install/rclrs_example_msgs/share/rclrs_example_msgs/rust/
install/rosgraph_msgs/share/rosgraph_msgs/rust/
install/statistics_msgs/share/statistics_msgs/rust/
install/std_msgs/share/std_msgs/rust/
install/test_msgs/share/test_msgs/rust/
install/unique_identifier_msgs/share/unique_identifier_msgs/rust/
```

```
ls -1 install/service_msgs/share/service_msgs

cmake
environment
hook
local_setup.bash
local_setup.dsv
local_setup.sh
local_setup.zsh
msg
package.bash
package.dsv
package.ps1
package.sh
package.xml
package.zsh
```

mxgrey, 2024-10-29:
I think the common message packages have been refactored, so we need to roll out a new `ros2_rust_jazzy.repos` file, and presumably we need to update `ros2_rust_rolling.repos`.

I'll test this out and open a PR.

mxgrey, 2024-10-30:
I've opened #425 to add a `ros2_rust_jazzy.repos` file.

Note that in the meantime I found that the `ros2_rust_rolling.repos` file still works when building against jazzy, so you can use that while the PR is waiting for review:

```
vcs import src < src/ros2_rust/ros2_rust_rolling.repos
```

azerupi, 2024-10-30:
That worked like a charm, thank you!
I tried copying over the humble repos file and change the versions to jazzy but I did not think about doing it from rolling. :slightly_smiling_face:

closed 2024-10-31 by azerupi