ros2-rust/ros2_rust #432: Spurious segmentation fault in Node::name doctest

mxgrey, 2024-11-21:
This [seg fault](https://github.com/ros2-rust/ros2_rust/actions/runs/11947109985/job/33302562685?pr=429) mysteriously happened during a run of CI for #429. I'm copying the error below in case the CI log gets erased in the future:

```
failures:

---- src/node.rs - node::Node::name (line 130) stdout ----
Test executable failed (signal: 11 (SIGSEGV) (core dumped)).

failures:
    src/node.rs - node::Node::name (line 130)

test result: FAILED. 25 passed; 1 failed; 0 ignored; 0 measured; 0 filtered out; finished in 5.41s

error: doctest failed, to rerun pass `--doc`
```

I don't see any clear reason why this would happen, but in the interest of emphasizing memory safety I don't think we should overlook it.

Maintainers can feel free to assign this issue ticket to me.

maspe36, 2025-02-14:
Found again, https://github.com/ros2-rust/ros2_rust/actions/runs/13328039205/job/37225589834?pr=443

nwn, 2025-02-23:
We may want to consider retaining core dumps from CI for easier debugging of these types of issues, like described here: https://github.com/itamarst/gha-upload-cores.