ros2-rust/ros2_rust #453: Issue compiling

RDaneelOlivav, 2025-01-15:
When compiling for humble and following the instructions for compilation it gives this error:

Starting >>> trajectory_msgs
--- stderr: rclrs
error[E0277]: the trait bound `rosidl_runtime_rs::String: From<Arc<str>>` is not satisfied
  --> src/parameter/service.rs:54:74
   |
54 |                         description: storage.options.description.clone().into(),
   |                                                                          ^^^^ the trait `From<Arc<str>>` is not implemented for `rosidl_runtime_rs::String`
   |
   = help: the trait `From<Arc<str>>` is not implemented for `rosidl_runtime_rs::String`
           but trait `From<&str>` is implemented for it
   = help: for that trait implementation, expected `&str`, found `Arc<str>`
   = note: required for `Arc<str>` to implement `Into<rosidl_runtime_rs::String>`
help: consider removing this method call, as the receiver has type `Arc<str>` and `Arc<str>: From<Arc<str>>` trivially holds
   |
54 -                         description: storage.options.description.clone().into(),
54 +                         description: storage.options.description.into(),
   |

error[E0277]: the trait bound `rosidl_runtime_rs::String: From<Arc<str>>` is not satisfied
  --> src/parameter/service.rs:55:85
   |
55 |                         additional_constraints: storage.options.constraints.clone().into(),
   |                                                                                     ^^^^ the trait `From<Arc<str>>` is not implemented for `rosidl_runtime_rs::String`
   |
   = help: the trait `From<Arc<str>>` is not implemented for `rosidl_runtime_rs::String`
           but trait `From<&str>` is implemented for it
   = help: for that trait implementation, expected `&str`, found `Arc<str>`
   = note: required for `Arc<str>` to implement `Into<rosidl_runtime_rs::String>`
help: consider removing this method call, as the receiver has type `Arc<str>` and `Arc<str>: From<Arc<str>>` trivially holds
   |
55 -                         additional_constraints: storage.options.constraints.clone().into(),
55 +                         additional_constraints: storage.options.constraints.into(),
   |

error[E0277]: the trait bound `rosidl_runtime_rs::String: From<Arc<str>>` is not satisfied
   --> src/parameter/service.rs:136:64
    |
136 |             let name: rosidl_runtime_rs::String = name.clone().into();
    |                                                                ^^^^ the trait `From<Arc<str>>` is not implemented for `rosidl_runtime_rs::String`
    |
    = help: the trait `From<Arc<str>>` is not implemented for `rosidl_runtime_rs::String`
            but trait `From<&str>` is implemented for it
    = help: for that trait implementation, expected `&str`, found `Arc<str>`
    = note: required for `Arc<str>` to implement `Into<rosidl_runtime_rs::String>`

error[E0277]: the trait bound `rosidl_runtime_rs::String: From<Arc<str>>` is not satisfied
   --> src/parameter/value.rs:345:33
    |
345 |                 string_value: v.into(),
    |                                 ^^^^ the trait `From<Arc<str>>` is not implemented for `rosidl_runtime_rs::String`
    |
    = help: the trait `From<Arc<str>>` is not implemented for `rosidl_runtime_rs::String`
            but trait `From<&str>` is implemented for it
    = help: for that trait implementation, expected `&str`, found `Arc<str>`
    = note: required for `Arc<str>` to implement `Into<rosidl_runtime_rs::String>`

error[E0277]: the trait bound `rosidl_runtime_rs::String: From<Arc<str>>` is not satisfied
   --> src/parameter/value.rs:370:64
    |
370 |                 string_array_value: v.iter().map(|v| v.clone().into()).collect(),
    |                                                                ^^^^ the trait `From<Arc<str>>` is not implemented for `rosidl_runtime_rs::String`
    |
    = help: the trait `From<Arc<str>>` is not implemented for `rosidl_runtime_rs::String`
            but trait `From<&str>` is implemented for it
    = help: for that trait implementation, expected `&str`, found `Arc<str>`
    = note: required for `Arc<str>` to implement `Into<rosidl_runtime_rs::String>`

For more information about this error, try `rustc --explain E0277`.
error: could not compile `rclrs` (lib) due to 5 previous errors
---
Failed   <<< rclrs [1min 35s, exited with code 1]
Aborted  <<< trajectory_msgs [1min 23s]
Aborted  <<< diagnostic_msgs [1min 28s]
Aborted  <<< shape_msgs [1min 29s]
Aborted  <<< nav_msgs [1min 48s]
Aborted  <<< sensor_msgs [2min 9s]

Has something changed? rustc 1.84.0 (9fc6b4312 2025-01-07)

mxgrey, 2025-01-16:
Rust 1.84 was released on 9 January. [This scheduled CI job](https://github.com/ros2-rust/ros2_rust/actions/runs/12760181616/job/35565209995) ran on 14 January, so I would expect it to be the 1.84. We should have the CI print out the rust version number just to double-check. 

There hasn't been a relevant commit to `rosidl_runtime_rs` nor `rclrs` in quite some time.

Did this happen with a completely fresh build?

RDaneelOlivav, 2025-01-16:
Yeah, It was with a fresh build in an Ubuntu 22 system . Here I'll leave the bash script that I used for installing the needed dependencies . And teh wird thing is thats several days ago it compiled correctly. 

```
  # Remove RUST to hav ethe latest version
  sudo apt remove rustc -y 
  sudo apt autoremove -y 
  curl --proto '=https' --tlsv1.2 -sSf https://sh.rustup.rs | sh -s -- -y
  
  # . "$HOME/.cargo/env"            # For sh/bash/zsh/ash/dash/pdksh
  # source "$HOME/.cargo/env.fish"  # For fish
  
  rustup update
  
  
  sudo apt update 
  sudo apt install ros-humble-test-msgs -y 
  
  cargo install cargo-ament-build
  sudo pip install git+https://github.com/colcon/colcon-cargo.git git+https://github.com/colcon/colcon-ros-cargo.git
  sudo apt install -y git libclang-dev python3-pip python3-vcstool
  cargo install --debug cargo-ament-build
  sudo apt install -y git libclang-dev python3-pip python3-vcstool # libclang-dev is required by bindgen
  # Install these plugins for cargo and colcon:
  sudo pip install git+https://github.com/colcon/colcon-cargo.git
  sudo pip install git+https://github.com/colcon/colcon-ros-cargo.git
  
  
  cd /home/user/rust_robot_ws/ros2_ws
  git clone https://github.com/ros2-rust/ros2_rust.git src/ros2_rust
  vcs import src < src/ros2_rust/ros2_rust_humble.repos
  . /opt/ros/humble/setup.sh
  colcon build
```

Am I missing something? Is there some tests of versions of similar that I can see where is the problem or any version that is sure to work?

mxgrey, 2025-01-16:
I created a fresh distrobox of Ubuntu 22.04 and followed your script (plus ROS 2 humble install instructions), and everything was built successfully, including rclrs and all the message bindings.

I've confirmed that I'm building with rust stable 1.84. I also tried building with the latest nightly, which is labeled 1.86-nightly, and that also compiled without any issues.

I'm afraid I really have no explanation for whatever you're experiencing. Without being able to reproduce the problem, I have absolutely no guess for what might be wrong. I'd start by completely deleting the colcon workspace (including `src`, `build`, and `install`) and trying over from scratch.

If that still doesn't help, then I might resort to clearing out the crate registry from cargo's cache (**note that this will require you to re-download the crate source code for all dependencies, but otherwise this should have no effect on your system**):

```bash
cargo install cargo-cache
```

```bash
cargo -r registry
```

RDaneelOlivav, 2025-01-16:
I executed both commands , the "cargo -r registry" game me this error:

`
    user:~/workspace$ cargo -r registry
    error: unexpected argument '-r' found
    
    Usage: cargo [+toolchain] [OPTIONS] [COMMAND]
           cargo [+toolchain] [OPTIONS] -Zscript <MANIFEST_RS> [ARGS]...
    
    For more information, try '--help'.
`

Could it be there is something in the system that is not allowing the correct compilation? Is there some way to be sure all teh system is clean and full proof to be able to do a successful build?

I can share with you the ROSject project that you can launch : [PROJECT](https://app.theconstruct.ai/l/67ed6e88/)

Thanks

mxgrey, 2025-01-16:
> the "cargo -r registry" game me this error:

Sorry there was a typo in the command. It was supposed to be: 

```bash
cargo cache -r registry
```

mxgrey, 2025-01-16:
> Could it be there is something in the system that is not allowing the correct compilation? Is there some way to be sure all teh system is clean and full proof to be able to do a successful build?

I have to assume this is somehow the case, but I honestly can't think of what it would be. The error is so peculiar that I can't think of what could lead to it.

RDaneelOlivav, 2025-01-17:
Hi Solved it by removing any previous system and being very carefull on teh sourcing. Thanks a lot for the help @mxgrey .

Here is the script I used for anybody with similar issue:

`
    #!/bin/bash
    
    # WARNING: THIS SCRIPT SHOULD BE RUN IN A VERY BIG INSTANCE (36 cores)
    
    # Step 1: Clean up any previous Rust installations
    echo "Cleaning up old Rust installations..."
    rm -rf ~/.cargo ~/.rustup
    sudo apt-get remove --purge rustc cargo -y
    sudo apt-get autoremove --purge -y
    sudo apt-get clean
    dpkg -l | grep rust
    sudo apt-get clean
    
    # Step 2: Install necessary dependencies
    echo "Installing necessary dependencies..."
    sudo apt-get update
    sudo apt-get install -y \
        curl \
        git \
        libclang-dev \
        python3-pip \
        python3-vcstool \
        sudo
    sudo rm -rf /var/lib/apt/lists/*
    
    sudo apt update
    sudo apt install ros-humble-test-msgs -y
    
    # Step 3: Install Rust
    echo "Installing Rust..."
    curl --proto '=https' --tlsv1.2 -sSf https://sh.rustup.rs | sh -s -- -y
    source "$HOME/.cargo/env"
    rustup update
    rustup default stable
    rustc --version
    cargo --version
    
    # Step 4: Install cargo-ament-build
    echo "Installing cargo-ament-build..."
    source "$HOME/.cargo/env"
    rustup update
    rustup default stable
    cargo install cargo-ament-build
    
    # Step 5: Install colcon packages
    echo "Installing colcon packages..."
    sudo pip install git+https://github.com/colcon/colcon-cargo.git git+https://github.com/colcon/colcon-ros-cargo.git
    
    # Step 6: Set up the workspace
    echo "Setting up the workspace..."
    mkdir -p /home/user/rust_ws/src
    cd /home/user/rust_ws/src
    
    # Clone the ros2_rust repo
    echo "Cloning the ros2_rust repository..."
    git clone https://github.com/ros2-rust/ros2_rust.git
    
    # Set the working directory back to the ROS2 workspace
    cd /home/user/rust_ws
    
    # Import ROS 2 dependencies
    echo "Importing ROS 2 dependencies..."
    vcs import src < src/ros2_rust/ros2_rust_humble.repos
    
    # Step 7: Build the workspace using colcon
    echo "Building the workspace..."
    source "$HOME/.cargo/env"
    rustup update
    rustup default stable
    rustc --version
    source /opt/ros/humble/setup.sh
    colcon build
    
    # Step 8: Set the Rust default for the user
    echo "Setting default Rust version for the user..."
    source /home/user/.cargo/env
    rustup default stable
    
    # Add Rust environment variables to .bashrc
    echo 'source $HOME/.cargo/env' >> /home/user/.bashrc
    echo 'rustup default stable' >> /home/user/.bashrc
    
    echo "Script execution completed successfully."
    
`

closed 2025-01-17 by RDaneelOlivav