ros2-rust/ros2_rust #458: rclrs 0.4.1 doesn't build for Rust 1.75

maspe36, 2025-02-08:
I am playing around in a workspace that doesn't clone ros2_rust but instead pulls from crates.io. I am trying to target Rust 1.75, our target version, and I am unable to build. 

I get the same issue as described in https://github.com/ros2-rust/ros2_rust/issues/449. In short, our the fix for this issue, https://github.com/ros2-rust/ros2_rust/pull/452 is not in rclrs 0.4.1.

maspe36, 2025-03-19:
Sure thing Julia, here you go

```
sam@orange:~$ rustup --version
rustup 1.27.1 (54dd3d00f 2024-04-24)
info: This is the version for the rustup toolchain manager, not the rustc compiler.
info: The currently active `rustc` version is `rustc 1.75.0 (82e1608df 2023-12-21)`
```

roboticswithjulia, 2025-03-19:
Hello @maspe36 I had same issue the other way and how I solved was uninstalling rust and install the newest version: 
> `rustup self uninstall`
> ` curl --proto '=https' --tlsv1.2 -sSf https://sh.rustup.rs | sh -s -- --default-toolchain 1.81.0 -y`  (maybe worth a try to install the last one 1.85.1 

Let me know if it solves the issue.

roboticswithjulia, 2025-03-20:
Yes, you are right! This is a better way to fix the issue rather than update rust version.  Let's leave this question to the maintainers of the repository then.

esteve, 2025-03-20:
@roboticswithjulia

> Yes, you are right! This is a better way to fix the issue rather than update rust version. Let's leave this question to the maintainers of the repository then.

FYI, @maspe36 is one of the maintainers of rclrs

roboticswithjulia, 2025-03-20:
Thank you @esteve for the clarification! Sorry, my bad! @maspe36 Apologies!! Then I was talking with the right person!!!

roboticswithjulia, 2025-03-20:
I was thinking to update the ./Dockerfile to use the last rust version, but  I am also bit unsure if this also would be aligned in how this MR should be fixed.

esteve, 2025-03-20:
@roboticswithjulia this ticket is because we officially support the 1.75 version of the Rust toolchain. When we hopefully release the first version of ros2-rust as part of a ROS distro, the version of Rust that Ubuntu ships with is 1.75, so that's why we don't want to update the version of Rust in Dockerfile either.

roboticswithjulia, 2025-03-20:
I understand now, thank you for the clarification @esteve !!

roboticswithjulia, 2025-03-25:
I do not know if this might also be related, but, when I am trying to build just the minimal_pub_sub examples, it neither cannot find the std_msgs . 
```
julia@jmr:~/Desktop/ros2_rust/examples/minimal_pub_sub$ cargo build
    Updating crates.io index
error: failed to select a version for the requirement `std_msgs = "*"`
  version 4.2.3 is yanked
location searched: crates.io index
required by package `examples_rclrs_minimal_pub_sub v0.4.1 (/home/julia/Desktop/ros2_rust/examples/minimal_pub_sub)`
```

roboticswithjulia, 2025-03-25:
⏫ Solved using the dockerfile https://github.com/ros2-rust/ros2_rust/blob/main/docs/building.md#option-2-using-the-docker-image instructions.