ros2-rust/ros2_rust #470: What's going on with rosidl_generator_rs?

hikari20XX, 2025-04-06:
Hello!

Sorry if this is a dumb question, but what's going on with rosidl_generator_rs? I've been working creating a bridge for the Blue Robotics Flight Controller (API is written in Rust) for some time now, and while I have a working version using standard message types, I'd really like to implement custom message types to avoid issues with string parsing down the road. However, whenever I try to include rosidl_generator_rs for type support functionality, it can't find the crate and the build fails. Tried following the instructions at the end of the building.md by sourcing the install directory, but no dice. Digging into ~/workspace/install/rosidl_generator_rs/share/rosidl_generator_rs, I can see it's missing the /rust directory (which I was looking for to hard code the path into the Cargo.toml in my custom_msgs directory).

Tl;dr: what's going on with rosidl_generator_rs? Am I just dumb? The basic pub/sub example works for me just fine. If anyone can offer any advice/wisdom, I'd really appreciate it.

mxgrey, 2025-04-07:
Although what you're describing sounds slightly different than what I experienced. 

If you do have `rosidl_generator_rs` in your workspace, then maybe double-check whether you've installed **both** colcon-cargo **and** colcon-ros-cargo.

hikari20XX, 2025-04-07:
Thank you so much, I really appreciate the help. I did double check both colcon-cargo and colcon-ros-cargo were installed (they were), and I tried doing the vcs import again, still to no avail. I'm able to use other bindings fine and build ROS2 packages with cargo just fine, so long as I don't include rosidl_generator_rs.

mxgrey, 2025-04-07:
When you say

> so long as I don't include rosidl_generator_rs.

Do you mean that inside your Cargo.toml you're doing something like this?

```toml
[dependencies]
rosidl_generator_rs = "*"
```

If that's what you're doing then you should remove that. `rosidl_generator_rs` is a Python package, not a Rust crate, so cargo doesn't care about it and can't use it.

In fact for your custom message package you shouldn't be using cargo at all. You should just use the typical [cmake and package.xml](https://docs.ros.org/en/rolling/Tutorials/Beginner-Client-Libraries/Single-Package-Define-And-Use-Interface.html#build-a-msg-file) files. As long as you have `rosidl_generator_rs` in the same workspace as your custom message package, the Rust bindings should get generated.

hikari20XX, 2025-04-07:
That is precisely what I was doing. I'm not sure when, but some point awhile ago I stopped using colcon and have just been using cargo. I'll give it another go! Thank you! :)

closed 2025-05-31 by maspe36