ros2-rust/ros2_rust #472: std_msgs version 4.2.3 is yanked

George-Gi, 2025-04-11:
Hi, I noticed that std_msgs crate is yanked.
```
error: failed to select a version for the requirement `std_msgs = "*"`
  version 4.2.3 is yanked
location searched: crates.io index

```
what is the suggested solution to this?

maspe36, 2025-04-11:
Hi @George-Gi, this is due to our complex message generation and consumption pipeline. When you put `std_msgs = "*"` in your `Cargo.toml`, there should be an accompanying `.cargo/config.toml` file in the root of your colcon workspace which "[patches](https://doc.rust-lang.org/cargo/reference/overriding-dependencies.html#the-patch-section)" any dependency on a `std_msgs` crate with some local crate. The local crate we patch to, should be generated by the [rosidl_generator_rs](https://github.com/ros2-rust/rosidl_rust/tree/main/rosidl_generator_rs) package when you `colcon build` within a workspace which also has an interfaces package containing `.msg` etc files.

So if you're seeing this issue, it _generally_ means one of these things
1. You may not have built with colcon yet.
Does the `.cargo/` folder exist in the root of your workspace?

2. You may not have the `rosidl_generator_rs` and/or `std_msgs` package(s) in your workspace or a sourced overlay
Does the `.cargo/config.toml` file exist but there are no entries?

---

I recognize this is really quite confusing, you're not the first person to bring this up and you won't be the last. There is an ongoing community discussion on how we might simplify this experience (https://github.com/ros2-rust/ros2_rust/discussions/471) if you're interested.

George-Gi, 2025-04-14:
> Hi [@George-Gi](https://github.com/George-Gi), this is due to our complex message generation and consumption pipeline. When you put `std_msgs = "*"` in your `Cargo.toml`, there should be an [...]

Hi @maspe36 , thank you for the reply. I didn't realize from the instructions  that it is necessary to install the generator and the rest of the repo. I thought this was optional only for development purposes and that by including only the relevant dependency in the Cargo.toml should do the job. So, cloning the whole repo is mandatory, right? My intention was to create the smallest possible container, that's why I want to avoid any extra stuff, not required.

maspe36, 2025-05-08:
You don't need to clone the whole ros2_rust repo, you can pull `rclrs` from crates.io via your crates Cargo.toml. But at the moment, the `rosidl_rust` package (which includes `rosidl_generator_rs`) needs to be in some underlay or the current colcon workspace.

We are in a weird transitional period where `rosidl_rust` is working its way into the rolling release[1]. Soon it should be sufficient to just pull `rclrs` from crates.io to write Rust-y ROS 2 nodes.

---
1. > Formal Decision: Add rosidl_rust to REP 2005 (ROS Project Charter - article 4.9.ii)
    > 
    > Motion: [@wjwwood](https://discourse.ros.org/u/wjwwood), Second: [@ahcorde](https://discourse.ros.org/u/ahcorde), passed by consensus 

https://discourse.ros.org/t/ros-pmc-minutes-for-april-22-2025/43346

dolezvo1, 2026-09-18:
I'm on ROS 2 jazzy and I seem to have the same issue, at least as far as I can tell.

```
--- stderr: my_ros_node                                                                                                                                      
error: failed to select a version for the requirement `std_msgs = "*"`
  version 4.2.3 is yanked
location searched: crates.io index
required by package `my_ros_node v0.1.0 (/home/ubuntu/src/my_ros_node)`
---
```

I have followed the steps for jazzy in the README (I have also additionally cloned the `test_interface_files` repo to fix https://github.com/ros2-rust/ros2_rust/issues/616), so I think I have the generator installed correctly, but I'm not 100 % sure:

```
- ws/
  - src/
    - common_interfaces/
    - my_ros_node/
    - example_interfaces/
    - examples/
    - rcl_interfaces/
    - rosidl_core/
    - rosidl_defaults/
    - rosidl_rust/
    - test_interface_files/
    - unique_identifier_msgs/
```

My `.cargo/config.toml` file exists, but does not seem to include any generated crates: 

```
[patch.crates-io]

```

Are there any other debugging steps I could try in order to get my node running?

maspe36, 2026-09-23:
Hey thanks for trying out rclrs :) 

The issue you are seeing @dolezvo1 is not the same, but manifests the same. We are in-between releases right now.

We're trying to release rclrs 0.8 (https://github.com/ros2-rust/ros2_rust/pull/595), which includes the full [ros-env](https://github.com/ros2-rust/ros-env) fix for pulling in the generated code from rosidl_generator_rs. I am assuming you're on rclrs 0.7, and to get around this today you could clone the ros2_rust repo in your workspace.

The majority of our efforts right now are towards making rclrs operate as a more "normal" Cargo crate. We discuss this in our Element chat or in our monthly community meetings if you're interested!

Also, since the issue you're seeing is different, if cloning the ros2_rust in your workspace doesn't work, please go ahead and open a new issue.

closed 2026-09-23 by maspe36