ros2-rust/ros2_rust #475: Creating subscriber after spin doesn't work

George-Gi, 2025-05-08:
If I haven't created any subscribers before spin and after spin I create a subscriber triggered by a callback, the subscriber doesn't work, i.e. the subscriber callback is never executed.
If I create a random subscriber before spin and then create the desired subscriber after spin, the callback of the desired subscriber works. I googled a bit around and I found this issue for ROS2, but it looked quite old and probably has been fixed.
Any ideas on that?

jhdcs, 2025-05-08:
May I ask why you're creating a subscriber after spinning the node? This feels like an unusual design pattern to me - usually your node has all desired/needed subscribers/publishers/clients/services already created before spinning.

George-Gi, 2025-05-08:
> May I ask why you're creating a subscriber after spinning the node? This feels like an unusual design pattern to me - usually your node has all desired/needed subscribers/publishers/clients/services already created before spinning.

We have a certain infrastructure so that the topics for some subscribers are set on the fly with a service

jhdcs, 2025-05-08:
> Creating subscribers on the fly will work after [#446](https://github.com/ros2-rust/ros2_rust/pull/446) is merged, but that's currently blocked on [#430](https://github.com/ros2-rust/ros2_rust/pull/430) getting approved.

I knew I was forgetting about something. My apologies. Review has been submitted.