ros2-rust/ros2_rust #477: Link against ROS crates for a new ROS node

George-Gi, 2025-05-12:
I have created a new ROS node and I have declared its dependencies in its Cargo.toml. If I do a symlink of my nodes `.cargo` to the `/ros2_ws/.cargo`  cargo build works fine. However, if I don't do this symlink, the ROS crates can't be found or I have to explicitly declare their path like `sensor_msgs = { version = "5.3.6", path = "/workspace/ros2_ws/install/sensor_msgs/share/sensor_msgs/rust" }`.
Is there another method that I am missing? I want to avoid the overhead of explicitly adding the paths and I still need the .cargo of my project to declare some env variables in the config.toml

romainreignier, 2025-05-12:
You are missing a colcon tool. 
Please install all the requirements mentionned in the README.

George-Gi, 2025-05-13:
> You are missing a colcon tool. Please install all the requirements mentionned in the README.

Thank you for the answer. I have followed the instructions of the Readme, could you please indicate which exactly tool is missing so that I can verify? I

romainreignier, 2025-05-13:
I don't know which one is missing.

closed 2026-09-23 by maspe36