ros2-rust/ros2_rust #479: Use thiserror for RclrsError

mxgrey, 2025-05-14:
Currently we're implementing `Error` for `RclrsError` manually which isn't best practice. The [`thiserror`](https://docs.rs/thiserror/latest/thiserror/) crate makes it very convenient to implement the standard `Error` trait and helps keep the implementation correct.

I'm opening this issue as a self-reminder, maintainers can feel free to assign this ticket to me.

alluring-mushroom, 2025-05-19:
I was going to do this after an earlier PR, but I then found https://github.com/dtolnay/thiserror/issues/245. I quite liked having the doccomments, and I meant to investigate further but never did.

I did another quick search just now and found this: https://github.com/dtolnay/thiserror/issues/44. That library looks interesting, but would be yet another dependency

mxgrey, 2025-05-20:
Interesting... I do like the idea of not duplicating the doc comments and the error attribute. The [example](https://github.com/yaahc/displaydoc?tab=readme-ov-file#example) that combines `displaydoc` and `thiserror` looks very sensible.

I'll give that a try when I get around to tackling this issue. Or you're very welcome to make a PR if that's something you were already planning to do, since it may be a while yet before I can get around to this.