ros2-rust/ros2_rust #487: Tutorial out of date

arjo129, 2025-06-04:
I was just following the following tutorial: https://github.com/ros2-rust/ros2_rust/blob/main/docs/writing-your-first-rclrs-node.md and discovered that the APIs have changed significantly since the last update of this tutorial.

Guelakais, 2025-08-13:
I've looked at the [corresponding](https://github.com/ros2-rust/ros2_rust/blob/main/docs/writing-your-first-rclrs-node.md) document, and I could rewrite it if you'd like me to. However, I don't think it can easily be adapted due to the API changes. Moreover, I think it would make more sense to add it directly to the source code. For example, you could write a 'documentation.rs' module and republish it through 'lib.rs' as follows:
```rust
mod documentation;
pub use documentation::*;
```
Therefore, it should be available through `docs.rs` at the next release and it wouldn't polute he `lib.rs` to much. Additionally, a functional example should be added to the [`examples`](https://github.com/ros2-rust/examples) repository, because people would want to try it.

esteve, 2025-11-23:
@vbalaji21 that's awesome, thanks! I've assigned it to you, feel free to submit a PR and we can have a look at it 🙂