ros2-rust/ros2_rust #500: import rclrs without building from source

ljburtz, 2025-08-06:
Thank you for the great software. We have a project that imports rclrs and it works well with the current ros2_rust main branch. 

To release our project more widely, it would help users if they could avoid needing to build rclrs from source (or go through the docker setup). Is there a timeline for the next release of rclrs ? 

0.4.0 from 2023 is the latest on both crates.io and this github's release page. 

idea: maybe a crate like `rclrs-rolling` could make the content of the main branch easier to install / try out (with explicitly no guarantee of stability)?

esteve, 2025-08-06:
@ljburtz thanks for using rclrs. We're about to make a new release of rclrs soon, I'm currently working on automating the release process so that we can make releases more often. On the other hand, some reasons why it took us so long is because of the API refactors and waiting for the code generator to officially be part of the ROS Kilted release (https://index.ros.org/p/rosidl_generator_rs/#kilted).

ljburtz, 2025-08-06:
@esteve great to hear! i know these things are hard but if you have any order of magnitude for when then maybe we can adjust our release to match

esteve, 2025-08-06:
@ljburtz if all goes well, maybe by the end of the week, but more realistically next week.

ljburtz, 2025-08-07:
that would match great with ours, thank you and good luck!

maspe36, 2025-08-22:
With the release of rclrs 0.5.0 (https://github.com/ros2-rust/ros2_rust/pull/51), assuming you're using the Kilted or Rolling distros, it should now be possible to have an rclrs node in a regular colcon workspace where rclrs is pulled from crates.io.